PaulStoffregen/SerialFlash #3: Sectorsizes

FrankBoesing, 2015-03-31:
You're using 64KB sectorsize, but i think better is 4KB (at least the W25Q128FV supports a 4KB sector erase)

PaulStoffregen, 2015-04-01:
There's 2 issues with 4K sector sizes.

1: Some larger chips don't support 4K sectors at all.

2: When 4K sectors are supported, erasing 64K by 16 sectors is dramatically slower then erasing a single 64K block.

Obviously 4K allows more efficient use of space when a large number of small erasable files need to be allocated.  Maybe we should add code to detect which chips support 4K sectors?  Then use a combination of 4K and 64K erase?  That would reduce the wasted SPI flash memory, but it would cost a lot of extra code and complexity in the flash memory on the Teensy.  Is that worthwhile?

FrankBoesing, 2015-04-01:
I would vote for 4K support. This is flash so i think we can espect that writes are not frequent. 
It really depends on the usage. Perhaps one could concatenate small files together if 64KB is a problem to get around the wasted memory. But ask others..

FrankBoesing, 2020-03-15:
Reopen (keep this as reference).