PaulStoffregen/SerialFlash #13: teensy 3.1 Audio 23lc1024

npashine, 2016-01-27:
Getting below error while performing read/write test of SRAM 23lc1024 with teensy 3.1.

ERROR---The flash chip may be left in an improper state.
                You might need to power cycle to return to normal.

Kindly help to resolve above.

Also tell how to connect audioplayserialflashraw to dac.

PaulStoffregen, 2016-01-27:
23LC1024 is a RAM chip, not Flash memory.  It's not supported by SerialFlash.

npashine, 2016-01-28:
Thanks paul..

But is there any way to create a file in ram than i can insert my data in
it and can link with audio connection DAC.

PaulStoffregen, 2016-01-28:
Please ask here:

https://forum.pjrc.com/forums/8-Audio-Projects

This issue tracker is for reporting bugs and discussing development with the library.

closed 2016-01-28 by PaulStoffregen