PaulStoffregen/SerialFlash #14: util/crc16.h not in Arduino Zero

get, 2016-02-03:
After obtaining the latest samd build from https://github.com/arduino/ArduinoCore-samd the arduino compiler complains:

```
 fatal error: util/crc16.h: No such file or directory
 #include "util/crc16.h"
```

Any suggestions?

PaulStoffregen, 2016-02-09:
Maybe try "stealing" the util/crc16.h from Teensy.

https://github.com/PaulStoffregen/cores/blob/master/teensy3/util/crc16.h

Or perhaps try upgrading from Due to Teensy 3.2....

closed 2016-02-09 by PaulStoffregen

get, 2016-02-13:
Thanks. I'm using Arduino Zero not Due, do you have any recommendations on how to modify this library in order to work with its ARM processor (atmel SAMD)? I've been having a bit of difficulty trying to do this and the issue seems to be that although the code works, it can't actually read or write to the memory.