PaulStoffregen/SerialFlash #16: Read/Write Clock Phases

drewrisinger, 2016-02-22:
I'm having an issue with the reads not clocking into a Teensy LC. It appears that the MISO pin is being driven low by the memory module after about 700-800ns, while the Teensy is between SPI.transfer() instructions. My best guess that the library was developed using the timings for a Teensy3, while the relatively long delay between transfers on a TeensyLC (with half the clock rate) is allowing the latch on the memory to relax/release down to 0. I'm using a Winbond W25Q128FV. 
Here's some scope screenshots. They are from the first few lines of the EraseEverything example program, with all lines hooked up correctly. D0 is connected to CS, D2 to MOSI, D3 to SCLK, D4 to MISO. The 9F command is the JEDEC "Read ID" command, and the first byte returning over MISO should be EF (for Winbond). I ran the same test several times, with one success (that success actually happened to be when I had the Vcc pin disconnected from 3.3V, somehow...)
[image]
[image]
[image]
[image]

drewrisinger, 2016-02-23:
When I set the Oscope to use the falling SCLK edge for the MISO signal, it works (which makes sense, because the memory is designed to read instructions on the rising clock edge, and send back on a falling edge).

As I see it, the **potential solutions** to this issue are:
- Speed up or reduce the library code that happens between transfers.

- Manually set the SPI control registers to read on falling edge, the CPOL (either using SPISettings or direct register writes)

- Make all commands some sort of variant on transfer16, which theoretically would have the clocked bytes transferred without any pause.

PaulStoffregen, 2016-02-23:
let's discuss this on the forum.
https://forum.pjrc.com/threads/33207-Can-t-read-Memory-ID-TeensyLC-amp-SerialFlash

PaulStoffregen, 2016-02-24:
Tested on known-good hardware, unable to reproduce problem.

https://forum.pjrc.com/threads/33207-Can-t-read-Memory-ID-TeensyLC-amp-SerialFlash?p=97405&viewfull=1#post97405

closed 2016-02-24 by PaulStoffregen