PaulStoffregen/SerialFlash #22: EraseEverything validation

manitou48, 2016-03-30:
It might be helpful if EraseEverything did some validation checks.  As it is, it says things are erased even if SPI is disconnected.  You might print the id[] and verify it's non-zero.  After erase, you could read page 0 and verify that it is 256  0xff's.  (of course, it could report 0xff's even if erase failed, but in my testing it seems to report 0's with disconnected/mis-wired SPI)

FrankBoesing, 2016-03-30:
Better is to read all pages, or parts of them. a) The erasing could be interrupted by the user (disconnects usb too early), b) maybe he does not use SerialFlash

FrankBoesing, 2016-04-28:
https://github.com/PaulStoffregen/SerialFlash/pull/26

returns capacity=0 for not-connected (or wrong cs etc) chips.

manitou48, 2018-01-07:
frank's pull fixed it

closed 2018-01-07 by manitou48