PaulStoffregen/SerialFlash #29: ESP8266

fredericplante, 2016-06-15:
+Hey Mister Stoffregen,
+
+I did not realy know how to reach you so Here it is :D
+
+Anyway, do you plan futur expension of your library to make it work with Arduino Due and ESP8266? 
+I know you might say that it already has support for the built on flash, on the ESP8266 case, but
+it would be nice ta have a bigger then then 4MB internal thing.
+
+Incase you have not eared of the ESP8266 device --> https://github.com/esp8266/Arduino <-- Worth the look!
+
+Any way, you library totally rull 2 thumbs up. :D

fredericplante, 2016-06-15:
I was reading around, I would aggrea to upgrade from Due to Tenzy. ;) I'm not to hot about what Arduino is becoming.

But I think it would be a great lost to set aside the ESP8266 board. 

Just saying :D

PaulStoffregen, 2016-06-15:
This library uses only the SPI library for hardware access.  Perhaps it can "just work" if you connect a serial flash with SPI?

I have not used any ESP board.  I am willing to accept contributions, but I personally do not plan to spend my own engineering time developing for the ESP chips.

closed 2016-06-15 by PaulStoffregen

fredericplante, 2016-06-15:
It's never that simple ;)

For those who are interested:

`Arduino: 1.6.5 (Windows XP), Board: "Generic ESP8266 Module, 80 MHz, 40MHz, DIO, 115200, 512K (64K SPIFFS), ck, Disabled, None"

Build options changed, rebuilding all

ListFiles.ino: In function 'void setup()':
ListFiles:31: error: no matching function for call to 'SerialFlashChip::readdir(char [64], unsigned int, long unsigned int&)'
ListFiles.ino:31:65: note: candidate is:
In file included from ListFiles.ino:1:0:
C:\Documents and Settings\gust\My Documents\Arduino\libraries\SerialFlash/SerialFlash.h:67:14: note: static bool SerialFlashChip::readdir(char*, uint32_t, uint32_t&)
  static bool readdir(char *filename, uint32_t strsize, uint32_t &filesize);
              ^
C:\Documents and Settings\gust\My Documents\Arduino\libraries\SerialFlash/SerialFlash.h:67:14: note:   no known conversion for argument 3 from 'long unsigned int' to 'uint32_t& {aka unsigned int&}'
no matching function for call to 'SerialFlashChip::readdir(char [64], unsigned int, long unsigned int&)'

  This report would have more information with
  "Show verbose output during compilation"
  enabled in File > Preferences.
`