PaulStoffregen/SerialFlash #31: Unknown chip (S25FL127S)

lensdigital, 2016-08-13:
I'm using S25FL127S (16MB) with ATMega1284p (SS is on pin D20 (PC4)  )
Erase Everything sketch works, takes about 40 seconds.
But when I run RawHardwareTest I get :
Read Chip Identification:
  JEDEC ID:     FF FF FF
  Part Nummber: (unknown chip)
  Memory Size:  0 bytes

PaulStoffregen, 2016-08-13:
FF FF FF means your hardware isn't connected properly.

lensdigital, 2016-08-13:
Strange, I just tried it for 10th time and it's working now. Must've needed to power-cycle it :)
Good news, this works with ATMega1284p running "maniacbug Mighty 1284p 16Mhz optiboot".

Raw SerialFlash Hardware Test

Read Chip Identification:
  JEDEC ID:     1 20 18
  Part Nummber: S25FL127S
  Memory Size:  16777216 bytes
  Block Size:   65536 bytes

Reading Chip...

Writing 8192 signatures

Double Checking All Signatures:
  all 8192 signatures read ok

Checking Signature Pairs
  all 4095 signature pairs read ok

Checking Read-While-Write (Program Suspend)
  write 256 bytes at 256
  write time was 540 microseconds.
  read-while-writing: 00 00 00 00 15 F5 95 4B 
  test passed, good read while writing

Checking Read-While-Erase (Erase Suspend)
  erase time was 150764 microseconds.
  erase correctly erased 65536 bytes
  read-while-erasing: 00 00 00 00 15 F5 95 4B 
  test passed, good read while erasing

closed 2016-08-13 by lensdigital