PaulStoffregen/SerialFlash #33: Issue with CopyFromSD/RawHardwareTest   using W25Q080 (adafruit 1564)

bmillier, 2016-09-21:
Both the eraseAll program and the rawhardwareTest program see the chip and identify it properly. The erase program says it has completed the erasing. But neither the RawhardwareTest nor CopyFromSD can write to the chip. The Write protect pin is Vcc as it should be. I am using an IC socket as this is a DIP part, and a W25Q128 (SOIC) in a DIP adapter works fine in the same socket. I tried several times and powered down as program suggested. I am pretty sure this is a good chip, (i.e. I don't think I screwed it up in the past). But, Oddly the data found in the chip looks suspiciously like AVR op codes, and I use AVRs a lot. Don't know why op codes would be there:  I might have used it to store data in an AVR project in the past but not code.

RawHardwareTest program output is:

Raw SerialFlash Hardware Test

Read Chip Identification:
  JEDEC ID:     EF 40 14
  Part Nummber: W25Q80BV
  Memory Size:  1048576 bytes
  Block Size:   65536 bytes

Reading Chip...
  Previous data found at address 0
  You must fully erase the chip before this test
  found this: 00 01 02 03 04 05 06 07 
     correct: 00 00 00 00 15 F5 95 4B 

Tests Failed  :{

The flash chip may be left in an improper state.
You might need to power cycle to return to normal.

npashine, 2016-10-12:
I'm facing the same problem as above.
Kindly give the solution as I'm in middle of very critical project which need serial flash working properly.

bmillier, 2016-10-12:
I took the easy route and used the Q128 version since I was using a socket. But I am pretty sure I had previously used the '080 with AVR MCU under Arduino. But, it was with a lower level EEPROM library, not the serialFlash library. If I get the chance I'll take another look to see if I can see what's wrong. I am guessing it is in the bulk erase function of the library, since the library can read/identify the '080 correctly.

FrankBoesing, 2016-10-12:
The W25Q80BV is not on the list of tested chips: https://github.com/PaulStoffregen/SerialFlash/blob/master/tests.txt

bmillier, 2016-10-12:
Well, it depends upon where you look, so I think we are both right. If you go up one level on Github, where the documentation for the library resides, it clearly shows the W25Q80BV as being compatible. That's what most "non-experts" will see first.
  They're all in the same Winbond family, so I suspect there is some small detail wrong in the SerialFlash library causing the issue.

PaulStoffregen, 2016-10-12:
The W25Q80BV was reported working.  I do not have one here for testing.

PaulStoffregen, 2016-10-12:
I tried to buy a W25Q80BV.  Digikey lists all 8 part numbers as obsolete and no stock.  Adafruit shows it as discontinued.

bmillier, 2016-10-12:
Yes, I realize that- I had stock from Adafruit while they still sold them. Not a problem for me, but it looked like a problem for npashine.

PaulStoffregen, 2016-10-12:
@npashine - Where are you buying the W25Q80BV chips?

FrankBoesing, 2016-10-12:
They are available @ Alieexpress

bmillier, 2016-10-12:
I just tried to run the SerialFlash eraseall and hardware test again on my '80BV chip. Same result- eraseall says it erased OK, but hardware test shows a pre-existing data pattern still present in chip i.e. not erased. Same board with the '128 in the socket works fine.

manitou48, 2016-10-19:
I just ran various tests on my Adafruit W25Q80BV.  All is still working for me.

FrankBoesing, 2016-10-19:
After i told "npashine" to connect WP with VCC, i've never got an answer  - so i think it works now.
Unfortunately, users like him don't give feedback.

bmillier, 2016-10-19:
@Frank. I don;t know about npashine, but WP was connected to Vcc on my board, and, using the same socket, the W25Q080 doesn't erase whereas a W25Q128 works fine.  But, I only have 1 '080 to try, so probably its a bad chip.

npashine, 2016-10-19:
Hi Frank,

I got screwed up in some real stuff and I didn't go with your scope.
wanna some time to put some test and will get back to you.

PaulStoffregen, 2016-10-19:
Indeed lack of feedback really diminishes our ability to improve the experience for everyone else.  :(

PaulStoffregen, 2016-10-19:
FWIW, I tried to buy some W25Q80BV chips from an Aliexpress vendor.  They claim to have shipped them last week.  Nothing has arrived yet.

When/if they show up, I'll add a W25Q80BV to my list of tested chips.

manitou48, 2016-10-19:
Actually, it's already on your list (README.md), I submitted data to you back in 2015.

PaulStoffregen, 2016-10-19:
Yeah, but when I get the actual chip, I'll put it on one of the test boards.  That way it'll be retested every time I work on this library... which admittedly hasn't been very often lately....

manitou48, 2016-10-19:
The Aliexpress chips look like Chinese knockoffs, i wonder if they under-reverse-engineered the  W25Q80BV.  Though the conversation suggests trouble with Adafruit DIP too?

npashine, 2016-10-25:
Hi all,

My things are working fine, probably problem was the long wires.
I made it AS(short)AP and started working fine.

PaulStoffregen, 2016-10-27:
@bmillier - My W25Q80BVs finally arrived.  Looks like mpashine's problem was resolved.  Are you still having trouble?

I have a *lot* of other software issues to investigate and features to add on various libraries, so if there isn't any remaining issue with W25Q80BV, can we close this issue and I'll put these chips into storage?  If there is still a problem and I should solder one of these chips to my test boards, please let me know.

npashine, 2016-10-27:
no, you focus on other things my issue got resolved.

bmillier, 2016-10-27:
Hi Paul:   Sorry about this. When I put W25Q128s in the same socket everything worked OK. I only had one '80 from Adafruit on hand, so I assumed the library had an error. But, obviously my one '80 was bad at least in the bulk erase (it read out the ID, info OK)   I've been in electronics for 40 years, and have to say that you, Robin at PJRC are the most dedicated people I've dealt with- to say nothing about your patience and expertise on all of the code you've written in support of Teensy. Never thought I'd abandon the AVR devices I've used for years.   My KS 3.5s arrived and tested fine. Now I need to find a good project for them.    Best regardsBrian Millier
Computer Interface Consultants
31 Three Brooks Dr.
Hubley N.S.
B3Z1A3
(902) 876-8645
Cell (902) 717-5834

closed 2016-10-27 by PaulStoffregen