PaulStoffregen/SerialFlash #36: New version of Microchip SST26VF0xxB

paulenuta, 2016-10-12:
Hi,
there is now a new version of Microchip SST26VF016B/032B/064B and BA which have new ID's BF, 26, [41, 42, 43] and also has a special command Global Block-Protection Unlock (ULBPR) [0x98] which has to be issued before any write/erase command beside the WREN [0x06]. I try to use RawHardwareTest.ino in which I added at line 436 this code `    if (id[1] == 0x26) {
      if (id[2] == 0x41) return "SST26VF016B";
      if (id[2] == 0x42) return "SST26VF032B";
      if (id[2] == 0x43) return "SST26VF064B";
    }` and then it recognize the chip but it did not made the tests. Also serial is not working unles I add `Serial.begin(115200);` at line 40 of the same sketch.
After Above change the output of the serial console is: `Raw SerialFlash Hardware Test

Read Chip Identification:
  JEDEC ID:     BF 26 42
  Part Nummber: SST26VF032B
  Memory Size:  `

Could you help with this chip?

Thanks a lot,
Paul

paulenuta, 2016-10-12:
Hi,
searching thru code I only find thin mention of manufacturer ID `#define ID0_SST		0xBF` but I did not find any use of this define.

Thanks a lot,
Paul

PaulStoffregen, 2016-10-22:
Is this the right part?
http://www.digikey.com/short/3bcfbr

paulenuta, 2016-10-24:
Right, B version has IOC defaulted to 0 -standard spi com and BA has IOC default 1 - quad spi.

paulenuta, 2016-10-24:
http://ww1.microchip.com/downloads/en/DeviceDoc/20005218E.pdf

PaulStoffregen, 2016-10-25:
I ordered a few of this Digikey part.  Should I have purchased a different one?

paulenuta, 2016-10-25:
The Digikey part you have ordered[SST26VF032B-104I/SM-ND] are OK, their default configuration is to standard SPI. I have tested SST26VF032BA-104I/SM [http://www.digikey.com/short/3b371z](url) also on standard SPI interface with and without setting IOC configuration bit to 0 and it was OK on my code (arduino, python and java on pi).
Be aware of special command Global Block-Protection Unlock (ULBPR) [0x98] that you need to issue after power on reset to be able to execute write operations (PAGE WRITE 0x02, SECTOR_ERASE = 0x20, CHIP_ERASE = 0xC7, etc).
So after you update the library I can also test on SST26VF032BA-104I/SM and confirm if it is ok.

> 
See http://ww1.microchip.com/downloads/en/DeviceDoc/20005218E.pdf, page 11, TABLE 4-3: CONFIGURATION REGISTER
>

FrankBoesing, 2016-10-25:
Hm. Where's the point of supporting this chip-variant, when there's a better version ?

paulenuta, 2016-10-25:
What do you mean by
> when there's a better version ?
are you referring to the quad spi version?

FrankBoesing, 2016-10-25:
Did'nt you write the [SST26VF032B-104I/SM-ND] are ok, and work ? Or did i mis-understand you (my english is not good)

paulenuta, 2016-10-25:
I meant they are OK for Paul to do the tests, they are not the same like the one I have, but the only difference I understood is the IOC bit default value, see his comment:

> I ordered a few of this Digikey part. Should I have purchased a different one?

PaulStoffregen, 2016-10-26:
>  they are not the same like the one I have

That would have been good information to write *before* I placed the Digikey order.

Which Digikey part number do I order to get *exactly* the same part you have?

paulenuta, 2016-10-26:
I have tested SST26VF032BA-104I/SM http://www.digikey.com/short/3b371z.
I tested only on standard SPI interface with and without setting IOC configuration bit to 0 and it was OK on my code (arduino, python and java on pi). The part you ordered and the one I have share same datasheet, you can verify on digikey or microchip sites.

paulenuta, 2016-10-27:
Hi Paul,
have you received the Digikey order?

PaulStoffregen, 2016-10-27:
I received the first chip, not the same one you're using.

paulenuta, 2016-11-02:
Hi Paul,
have you got the time to test the flash?
Thanks a lot,
Paul

PaulStoffregen, 2016-11-02:
No, not yet.  But I did at least get them soldered to a board... so it's on my long list of stuff to do....

PaulStoffregen, 2016-11-02:
I have another new board coming for 2 more chips.  Will probably wait 'til all these new chips are ready to test.

paulenuta, 2017-01-19:
Hello Paul, any update on this part support? If you don't have time for it we can close the issue.

closed 2017-05-13 by paulenuta