PaulStoffregen/SerialFlash #37: FS structure

Dragon-Knight, 2016-11-07:
Hello.
You couldn't on tell about structure of file system in more detail?
I have an idea to write the program for the computer that to program a chip of memory via the programmator.

FrankBoesing, 2016-11-07:
Did you try FrankBoesing/TeensyTransfer ?

Dragon-Knight, 2016-11-07:
I apologize, my knowledge of English very bad and probably I not correctly expressed...

I want to create the folder on the computer, to place in it all necessary files, to launch the program which I want to write and on an output to receive the RAW file which I will write in flash memory via the programmator then I will be able to use this library for operation with file system.

FrankBoesing, 2016-11-07:
Yes, you can do that with TeensyTransfer.

There's a *.exe  (and executables for MAC / LINU)  and a Sketch.

https://forum.pjrc.com/threads/33859-TeensyTransfer?p=101306

Dragon-Knight, 2016-11-07:
Thanks for the response.

But how I understood your library only for Teensy 3.x boards, i.e. for ARM microcontrollers?
My task to make for ATmega 328/644/1284 microcontrollers..., i.e. only AVR 8bit.

FrankBoesing, 2016-11-07:
Okay, you're right.

closed 2016-11-08 by PaulStoffregen

Dragon-Knight, 2016-11-08:
But....
OK, thanks!