PaulStoffregen/SerialFlash #38: SPI Class limitation in Simblee

biomurph, 2016-12-05:
When compiling for Simblee, there is an error in SerialFlashChip.cpp line 161:

`SPIPORT.transfer(p, rdlen);`

Simblee SPI Class does not have a transfer function that receives a buffer. I did this

```
for(int i=0; i<rdlen; i++){
	SPIPORT.transfer(p[i]);	//(p, rdlen);
}
```

nkings10, 2018-04-04:
Same issues with the Espressif 32 (ESP32). You quick fix lets me compile, still yet to check the library functions. Thanks @biomurph

rfestag, 2019-03-05:
@nkings10 - I am having issues using this library with ESP32. The above issue appears to no longer exist (it compiles out of the box), but successive reads/writes don't appear to work. 

Specifically, during the check_signature() call, if the chip was recently flashed, it will attempt to write the magic number at the beginning, and then immediately read it back. This process fails (the data read back does not match what should have been written). I suspect it an issue in the bowels of arduino-esp32, but I'm not sure where or how to debug it.

nkings10, 2019-03-05:
@rfestag I ended up dropping the project i was using this library for. So i never got further with it.