PaulStoffregen/SerialFlash #43: Copy From SD to Flash

GumboBeans, 2017-06-07:
Hi, I have spent the last 2 days attempting to figure this out, and i'm lost, so hopefully someone here could help out. 

Is there a way of copying a file from the SD Card to the SPI Flash and place it at 0x00000000, I don't need to list the files or even know they exist on the flash, All that needs to be done it to copy a single file to the 0x00 position on the SPI Flash.

Any help would be amazing.

PaulStoffregen, 2017-06-07:
Yes.  You should use the SPIFlash library, instead of SerialFlash.  It gives raw data access to the chip, rather than trying to implement a file-like abstraction.

GumboBeans, 2017-06-08:
Thank you for the quick reply Paul, I shall look into that :)

closed 2017-06-12 by PaulStoffregen