PaulStoffregen/SerialFlash #46: Issue running RawHardwareTest code

haapeng, 2017-06-28:
Hi mate, I using spansion s25fl512s f.memory and run the RawHardwareTest code but the test failed.
I attached the results and my circuit here,
[testing2.pdf](https://github.com/PaulStoffregen/SerialFlash/files/1108769/testing2.pdf)

roy-zahor, 2019-01-22:
Same issue here when using W25Q80DV.
I get the exact same results (even the byte count is the same).

PaulStoffregen, 2019-01-22:
This is the wrong place to ask for help.  Post your questions on the Arduino forum.

When you ask on the forum, show clear photos of how you've actually connected the hardware.  Nobody can help when you give no useful info.

Again, this issue tracker is *not* the place to ask for project help.  Ask on the forum.

closed 2019-01-22 by PaulStoffregen