PaulStoffregen/SerialFlash #49: Identification and reading of Micron N25Q064A

hackwin, 2017-09-18:
I'm having trouble with this library to identify a N25Q064A chip by Micron.  Is it compatible?

`#include <SerialFlash.h>
#include <SPI.h>

const int FlashChipSelect = 6; // digital pin for flash chip CS pin

void setup() {

  Serial.begin(115200);

  while (!Serial) ;
  delay(100);

  Serial.println("Raw SerialFlash Hardware Test");
  SerialFlash.begin(FlashChipSelect);

  unsigned char buf[1];
  
  Serial.println();
  Serial.println("Reading Chip...");
  
  for(unsigned long i=0; i< 4294967295L; i++){ 
    Serial.print("  addr = ");
    Serial.print(i, HEX);
    Serial.print(", data = ");
    SerialFlash.read(i, buf, 1);
    Serial.print(buf[1]);
    Serial.println();
  }
}

void loop() {
  // do nothing after the test
}`

PaulStoffregen, 2017-09-18:
https://forum.pjrc.com/threads/46474-Read-Micron-N25Q064A-Flash-Chip-using-SerialFlash-library

Do not open an issue here until the problem is confirmed to be a bug in this library.

closed 2017-09-18 by PaulStoffregen

tolson2000, 2017-09-18:
Shouldn't it be 4294967295UL not  4294967295L

On 09/18/2017 09:06 AM, hackwin wrote:
> for(unsigned long i=0; i< 4294967295L; i++){