PaulStoffregen/SerialFlash #58: WINBOND W25M02GV serial NAND support

fa1ke5, 2019-03-05:
Hi, im yousing WINBOND W25M02GV serial NAND for my project on esp32
if you want, i make some tests for support this chip in you library.
in this time i take JEDEC ID on SPI slow speed 20-26mHz
Raw SerialFlash Hardware Test

Read Chip Identification:
  JEDEC ID:     0 EF AB A5
  Part Nummber: (unknown chip)
  Memory Size:  1048576 bytes
  Block Size:   65536 bytes

Reading Chip...
  Previous data found at address 0
  You must fully erase the chip before this test
  found this: 00 00 00 00 00 00 15 F5 
     correct: 00 00 00 00 15 F5 95 4B 

Tests Failed  :{

The flash chip may be left in an improper state.
You might need to power cycle to return to normal.

As you can see 1st byte in JEDEC ID is dummy.
2nd and 3d bytes is okey, but 4th byte is wrong.
On datsheet doc i can see correct JEDEC ID as EF AB 21.

I'm fork you repo and may be make some changes from you code.
Regards....

PaulStoffregen, 2019-03-05:
Looks like SPI communication is unreliable.  Maybe related to #57.

fa1ke5, 2019-03-05:
But on SPI speed 26Mhz or less communication is normal on mode 0
I'm using standard code for spi exchange and now I'm getting correct ID 00 EF AB 21, and I see in you sources wrong logic for reading ID of this chip.

PaulStoffregen, 2019-03-05:
Any chance you could be more specific about the "wrong logic"?

This code has worked across many different boards.  That doesn't mean it's bug-free, but such a claim really needed to be back by specific details.

FrankBoesing, 2019-03-05:
For this specific chip, he's right. The datasheet says the chip needs 8 dummyclocks after cmd 9Fh.
Well, the chip is not in your compatibility list. It is 2x 1G-Bit.

FrankBoesing, 2019-03-05:
It needs a "Software Die Select (C2h)" cmd - interesting. Is has one chipselect and switches the dies via software.

fa1ke5, 2019-03-05:
> For this specific chip, he's right. The datasheet says the chip needs 8 dummyclocks after cmd 9Fh.
> Well, the chip is not in your compatibility list. It is 2x 1G-Bit.

8 dummyclocks = 1 byte of SPI and now we are reading 00 in first SPI byte and real ID start transferring in second byte, but 3d byte you use for somemore and we are have not space for reading full 3 byte chip ID.

> It needs a "Software Die Select (C2h)" cmd - interesting. Is has one chipselect and switches the dies via software.

in one case we have two crystal on one SPI bus and C2h used for software chip select.

FrankBoesing, 2019-03-05:
Well, other chips don't want these dummyclocks. That's solvable (->If ChipID==0 read one more byte). But more important is the C2h command.

fa1ke5, 2019-03-05:
> Well, other chips don't want these dummyclocks. That's solvable (->If ChipID==0 read one more byte). 
i think about this, but how do we distinguish 00 from the answer of broken chip?

fa1ke5, 2019-03-05:
But more important is the C2h command.

As i know its full analogue of two independent chips on one SPI bus.

fa1ke5, 2019-03-05:
....and one more, i dont find chip-erase command in documentation. only erase block

FrankBoesing, 2019-03-05:
Where did you buy that chip?

fa1ke5, 2019-03-05:
in Russia https://www.compel.ru/infosheet/WINBOND/W25M02GVZEIG%20TR

fa1ke5, 2019-03-05:
W25N04KV is more interesting, but i dont find him in stock (:
http://www.spiflash.com/hq/product/code-storage-flash-memory/serial-nand-flash/index.html?__locale=ja&partNo=W25N04KV

fa1ke5, 2019-03-18:
Hi all!
I make some changings from original lib and now it can read, write, erase this chip in both DIEs all 256MB
But still need some work for adaptate it for work wiht other stuff...
You can see changed lib in my repo
https://github.com/fa1ke5/SerialFlash/blob/master/SerialFlashChip.cpp

fa1ke5, 2019-03-18:
Raw SerialFlash Hardware Test
Software resetting flash chip
Found Winbond NAND flash 1, unlocking ...
Found Winbond NAND flash 2, unlocking ...

Read Chip Identification:
  JEDEC ID:     EF AB 21
  Part Nummber: W25M02GVZEIG
  Memory Size:  268435456 bytes
  Block Size:   131072 bytes

Reading Chip...
Current DIE = 0
need DIE = 1 changing..
Writing 524288 signatures

FrankBoesing, 2019-03-20:
The ID for the IPS6404 SPIRAM is even more weird: Repsonse is 3xrandom bytes (I saw 0xff , 0xfe and 0x00 randomly), then in the folllowing bytes the real ID: 0x0d, 0x5d

mansoorzamankhan, 2023-11-16:
> Hi all! I make some changings from original lib and now it can read, write, erase this chip in both DIEs all 256MB But still need some work for adaptate it for work wiht other stuff... You can see changed lib in my repo https://github.com/fa1ke5/SerialFlash/blob/master/SerialFlashChip.cpp

how can I change the library for "W25Q02JV (four x 512M-bit stack die) " ?

PaulStoffregen, 2023-11-16:
You're asking about a NOR flash chip on an issue originally about NAND flash chips.  The part numbers look similar, but those letters which differ are very important.  The way these NAND vs NOR chips work is very different.

Just to state as clearly as possible, NAND chips are not supported by this library and probably never will be.

Multiple die NOR chips also aren't supported.  Perhaps with substantial programming they could be supported.  But I don't have time to write and test that code, nor to write a detailed message or otherwise guide you to writing that code.

mansoorzamankhan, 2023-11-17:
> Hi all! I make some changings from original lib and now it can read, write, erase this chip in both DIEs all 256MB But still need some work for adaptate it for work wiht other stuff... You can see changed lib in my repo https://github.com/fa1ke5/SerialFlash/blob/master/SerialFlashChip.cpp

can you guide me on how to modify this library for a 256MB (4 dies) NAND flash.

fa1ke5, 2023-11-17:
You need control variable act_die in my code, and switch it according column address, see to strings num about 145 on my code.

fa1ke5, 2023-11-17:
My english is so poor, sorry.
Я модифицировал код библиотеки что бы последовательно писать и читать блоки данных в все DIE сразу. Логика состоит в следующем - операции очистки и записи занимают длительное время, поэтому мы заполняем кэш и параллельно готовим остальные DIE для записи следующих данных. То есть пока мы птшем данные в die0 мы стираем блок в die1 и так далее. Вам нужно всего лишь исправить адреса колонн и количество die для вашей версии чипа

mansoorzamankhan, 2023-11-20:
> You need control variable act_die in my code, and switch it according column address, see to strings num about 145 on my code.

I am using NOR flash W25Q02JV-DTR with 4 dies
https://www.winbond.com/hq/support/documentation/levelOne.jsp?__locale=en&DocNo=DA00-W25Q02JV
 how can I find the column addresses?

closed 2023-12-12 by PaulStoffregen