PaulStoffregen/SerialFlash #59: cFeon q32b-104hip EN25Q32B support

EUA, 2019-03-09:
Simply not detected. I got this:
```
Read Chip Identification:
  JEDEC ID:     FF FF FF
  Part Nummber: (unknown chip)
  Memory Size:  0 bytes

Tests Failed  :{

The flash chip may be left in an improper state.
You might need to power cycle to return to normal.
```
I believe all SPI flashes are compatible with each other.

Here is datasheet.
What could be the problem?
Thanks.

PaulStoffregen, 2019-03-09:
> What could be the problem?

Hardware not wired properly.

EUA, 2019-03-09:
Nope, but not know If I need to add pullup to HOLD. Winbond chips work without pullup. But this one requires. Thanks.

closed 2019-03-09 by EUA

mauron85, 2019-09-25:
@EUA How did you resolved your problem? Sorry, but as newbie I don't know what pullup to HOLD means. Thanks.

EUA, 2019-09-25:
@mauron85 It work after just adding a resistor (let say 1k) between VCC and "HOLD" pin. That pulls voltage up to logic 1. in short, PullUp.

mauron85, 2019-09-26:
I was only using clippers as chip is still soldered to the router board, but it was not detected by CH341A. Will desolder it and try your advice. Thank you @EUA.

mauron85, 2019-09-26:
I've got last question, before I start desoldering. Can this flash chip be replaced with W25Q128FVSG? https://www.pjrc.com/teensy/W25Q128FV.pdf It will save me effort, if it's not compatible. Thanks and sorry for the hassle.

EUA, 2019-10-01:
Sure. I believe it will work %99.