PaulStoffregen/SerialFlash #60: ESP32 Broken

EUA, 2019-03-10:
Arduino 1.8.8 + SerialFlash 0.5 + esp32 1.0.0
Using with Arduino-ESP32, SerialFlash broken. Examples doesn't working. Gives this error.
```
/home/death/Arduino/libraries/SerialFlash/SerialFlashChip.cpp: In static member function 'static void SerialFlashChip::read(uint32_t, void*, uint32_t)':
/home/death/Arduino/libraries/SerialFlash/SerialFlashChip.cpp:162:21: error: invalid conversion from 'uint8_t* {aka unsigned char*}' to 'uint8_t {aka unsigned char}' [-fpermissive]
   SPIPORT.transfer(p);
                     ^
In file included from /home/death/Arduino/libraries/SerialFlash/SerialFlash.h:32:0,
                 from /home/death/Arduino/libraries/SerialFlash/SerialFlashChip.cpp:28:
/home/death/.arduino15/packages/esp32/hardware/esp32/1.0.0/libraries/SPI/src/SPI.h:69:13: note:   initializing argument 1 of 'uint8_t SPIClass::transfer(uint8_t)'
     uint8_t transfer(uint8_t data);
             ^
```

fa1ke5, 2019-03-10:
what code are you using?

EUA, 2019-03-11:
Examples provided with library.

fa1ke5, 2019-03-11:
I don't testing other examples, but rawhardwaretest compiled well on esp32

fa1ke5, 2019-03-11:
Why you are using old esp32 sdk, as I know current version of him much more than 1.0.0

EUA, 2019-03-12:
My arduino installation string just shows 1.0.
Well I updated it and error solved.
Thanks.

closed 2019-03-12 by EUA