PaulStoffregen/SerialFlash #65: Support for Adesto SPI Flash Chip

AKASHWORLD614, 2020-01-17:
i have seen your library and used it in my project will this support the flash chip i am willing to use or no 

https://lcsc.com/product-detail/FLASH_Adesto-Technologies-AT25SF321-SHD-T_C226100.html

if not then please say to me one with 32mb or more from website LCSC

FrankBoesing, 2020-01-17:
These are tested:

https://raw.githubusercontent.com/PaulStoffregen/SerialFlash/master/tests.txt

AKASHWORLD614, 2020-01-17:
what should replace size with in this code 

SerialFlash.createErasable(filename, size);

what is the difference between these two 
SerialFlash.create(filename, size);
                       &
SerialFlash.createErasable(filename, size);

can we expand size of file as data grows in erasable?

FrankBoesing, 2020-02-05:
For questions, please use the forum.

PaulStoffregen, 2020-02-05:
questions here:
https://forum.pjrc.com/

closed 2020-02-05 by PaulStoffregen