PaulStoffregen/SerialFlash #69: Incorrect capacity detection and strange behavior for W25Q256JV with Arduino Uno

swapnilsayansaha, 2020-08-03:
I connected a Winbond W25Q256JV to Arduino Uno. The example codes (e.g. RawHardwareTest) shows that the flash has 32 Mb of memory, which is not correct (the flash has eight 32 Mb sectors, totaling 256 Mb). 

Secondly, the example code (RawHardwareTest) gets stuck at:  
chipsize = SerialFlash.capacity(buf);

FrankBoesing, 2020-08-04:
A) 256Mbit are 32M Bytes. 
B) I don't understand. You say RawHardwareTest shows a size (32Mb - which is correct) ,and "secondly it gets stuck at SerialFlash.capacity(buf)".  If it would stuck, it could'nt display a size. Then, the code used is here: https://github.com/PaulStoffregen/SerialFlash/blob/master/SerialFlashChip.cpp#L452 - there is now way it gets stuck.

swapnilsayansaha, 2020-08-04:
Hi, thanks for pointing out. Sorry I got confused. The code gets stuck right after capacity is shown here:

```
Serial.println("Reading Chip...");
 memset(buf, 0, sizeof(buf));
 memset(sig, 0, sizeof(sig));
 memset(buf2, 0, sizeof(buf2));
```

nick-nikoliuk, 2021-02-24:
The same behavior here

nick-nikoliuk, 2021-02-24:
Ok, the problem is that lib overuse "strings" that stored in RAM. Arduino or other atmegas do not have enough memory for this.
Just wrap all the logs with F() like Serial.print(F("Your logs")). This will reduce load and test starts work like a charm.

PaulStoffregen, 2021-03-03:
How does this look?
https://github.com/PaulStoffregen/SerialFlash/commit/8c310813003388455b8738bddda34eb03f3a3ef3

Does it fully solve this issue?

FrankBoesing, 2021-03-03:
I don't think the MP3payer works with the Uno...

PaulStoffregen, 2021-03-03:
LOL

I'm going to assume this issue is resolved and close it now.  If anyone disagrees, just reply with details.

closed 2021-03-03 by PaulStoffregen

msaid2010, 2023-05-18:
hi I use Atmega128 and I made level shifting for W25Q64 when I tested the code ( RawHardwareTest ) it always fail 
[image]
is there any problem in the Hardware ???
or What the problem ?