PaulStoffregen/SerialFlash #75: Issues testing W25Q64 Breakout board

Harveyn4444, 2021-03-03:
### Description

Describe your problem.
I have a breakout board containing a W25Q64 flash chip and have tried to run the raw hardware test.

### Steps To Reproduce Problem

Please give detailed instructions needed for anyone to attempt to reproduce the problem.
I have the breakout board (https://www.ebay.co.uk/itm/Flash-Storage-Module-W25Q64-64M-bit-SPI-Interface-For-1-PACK/323636815798) found here, with the Do and Di pins set to MOSI and MISO respectively and the example codes cannot read the chip, with the CS pin at 6. 

### Hardware & Software

Teensy LC
(https://www.ebay.co.uk/itm/Flash-Storage-Module-W25Q64-64M-bit-SPI-Interface-For-1-PACK/323636815798) No idea. of the actual name
Arduino IDE 1.8.13
Teensyduino version 1.53
Version info & package name (from Tools > Boards > Board Manager)
macOS Big Sur version 11.1
### Arduino Sketch

```cpp
// Change the code below by your sketch (please try to give the smallest code which demonstrates the problem)

//This is the example code 
#include <SerialFlash.h>
#include <SPI.h>

const int FlashChipSelect = 6; // digital pin for flash chip CS pin
//const int FlashChipSelect = 21; // Arduino 101 built-in SPI Flash

SerialFlashFile file;

const unsigned long testIncrement = 4096;

void setup() {

  //uncomment these if using Teensy audio shield
  //SPI.setSCK(14);  // Audio shield has SCK on pin 14
  //SPI.setMOSI(7);  // Audio shield has MOSI on pin 7

  //uncomment these if you have other SPI chips connected
  //to keep them disabled while using only SerialFlash
  //pinMode(4, INPUT_PULLUP);
  //pinMode(10, INPUT_PULLUP);

  Serial.begin(9600);

  while (!Serial) ;
  delay(100);

  Serial.println("Raw SerialFlash Hardware Test");
  SerialFlash.begin(FlashChipSelect); // proceed even if begin() fails

  if (test()) {
    Serial.println();
    Serial.println("All Tests Passed  :-)");
    Serial.println();
    Serial.println("Test data was written to your chip.  You must run");
    Serial.println("EraseEverything before using this chip for files.");
  } else {
    Serial.println();
    Serial.println("Tests Failed  :{");
    Serial.println();
    Serial.println("The flash chip may be left in an improper state.");
    Serial.println("You might need to power cycle to return to normal.");
  }
}

bool test() {
  unsigned char buf[256], sig[256], buf2[8];
[... 404 lines cut ...]
}

bool is_erased(const unsigned char *data, unsigned int len)
{
	while (len > 0) {
		if (*data++ != 255) return false;
		len = len - 1;
	}
	return true;
}

void printbuf(const void *buf, uint32_t len)
{
  const uint8_t *p = (const uint8_t *)buf;
  do {
    unsigned char b = *p++;
    Serial.print(b >> 4, HEX);
    Serial.print(b & 15, HEX);
    //Serial.printf("%02X", *p++);
    Serial.print(" ");
  } while (--len > 0);
  Serial.println();
}

Thank you

PaulStoffregen, 2021-03-03:
DO connects to MISO and DI connects to MOSI

Harveyn4444, 2021-03-03:
Yep literally fixed it straight away, sorry for being stupid :)

closed 2021-03-03 by Harveyn4444

Harveyn4444, 2021-03-04:
The exact same code doesn't work with my teensy 4.0, and this time is correctly wired up.
I've just tried it on my teensy LC and it works, just not on my Teensy 4.0, nothings changed, just the board.
Any ideas Thank you again

PaulStoffregen, 2021-03-05:
Time to start a forum thread.  Please show a photo of the wiring when you post on the forum.