PaulStoffregen/SerialFlash #76: Teensy 4.0 and w25Q64 Breakout Board

Harveyn4444, 2021-03-05:
Description

Describe your problem.
I have a breakout board containing a W25Q64 flash chip and have tried to run the raw hardware test.

Steps To Reproduce Problem

Please give detailed instructions needed for anyone to attempt to reproduce the problem.
I have the breakout board (https://www.ebay.co.uk/itm/Flash-Storage-Module-W25Q64-64M-bit-SPI-Interface-For-1-PACK/323636815798) found here, with the Do and Di pins set to MISO and MOSO respectively and the example codes cannot read the chip, with the CS pin at 4.

Hardware & Software

Teensy 4.0
(https://www.ebay.co.uk/itm/Flash-Storage-Module-W25Q64-64M-bit-SPI-Interface-For-1-PACK/323636815798) No idea. of the actual name
Arduino IDE 1.8.13
Teensyduino version 1.53
Version info & package name (from Tools > Boards > Board Manager)
macOS Big Sur version 11.1

Arduino Sketch

//This is the example code 
#include <SerialFlash.h>
#include <SPI.h>

const int FlashChipSelect = 4; // digital pin for flash chip CS pin
//const int FlashChipSelect = 21; // Arduino 101 built-in SPI Flash

SerialFlashFile file;

const unsigned long testIncrement = 4096;

void setup() {

  //uncomment these if using Teensy audio shield
  //SPI.setSCK(14);  // Audio shield has SCK on pin 14
  //SPI.setMOSI(7);  // Audio shield has MOSI on pin 7

  //uncomment these if you have other SPI chips connected
  //to keep them disabled while using only SerialFlash
  //pinMode(4, INPUT_PULLUP);
  //pinMode(10, INPUT_PULLUP);

  Serial.begin(9600);

  while (!Serial) ;
  delay(100);

  Serial.println("Raw SerialFlash Hardware Test");
  SerialFlash.begin(FlashChipSelect); // proceed even if begin() fails

  if (test()) {
    Serial.println();
    Serial.println("All Tests Passed  :-)");
    Serial.println();
    Serial.println("Test data was written to your chip.  You must run");
    Serial.println("EraseEverything before using this chip for files.");
  } else {
    Serial.println();
    Serial.println("Tests Failed  :{");
    Serial.println();
    Serial.println("The flash chip may be left in an improper state.");
    Serial.println("You might need to power cycle to return to normal.");
  }
}

bool test() {
  unsigned char buf[256], sig[256], buf2[8];
  unsigned long address, count, chipsize, blocksize;
  unsigned long usec;
[... 411 lines cut ...]
}

void printbuf(const void *buf, uint32_t len)
{
  const uint8_t *p = (const uint8_t *)buf;
  do {
    unsigned char b = *p++;
    Serial.print(b >> 4, HEX);
    Serial.print(b & 15, HEX);
    //Serial.printf("%02X", *p++);
    Serial.print(" ");
  } while (--len > 0);
  Serial.println();
}

Thank you

Picture included
Red - Vcc (3.3V)
Yellow - CS (4)
Green - DO MISO Pin(12)
Brown - GND
Purple - SCK Pin(13)
Blue - DI MOSI Pin(11)
[image]

Harveyn4444, 2021-03-05:
After re uploading the example sketch the program can now read the chip, just after I posted this :), but over the last few days it would not read the chip, ID the chip and not read it but now can read the chip and passes the test, I wired the chip up the exact same every time and just manages to work after I posted the issue.

Harveyn4444, 2021-03-05:
I've just unplugged a rewired everything again and get this message 
<Raw SerialFlash Hardware Test

Read Chip Identification:
  JEDEC ID:     FF FF FF
  Part Nummber: (unknown chip)
  Memory Size:  0 bytes

Tests Failed  :{

The flash chip may be left in an improper state.
You might need to power cycle to return to normal.>

Harveyn4444, 2021-03-19:
Also when I try the erase all example it freezes my serial port and outputs nothing, if I unplug the teensy the text is immediately written to the serial output
Again the chip seemed to be able to be read by my Teensy LC button. my Teensy 4.0 as I've just been trying again, no idea what could be causing the issue

Harveyn4444, 2021-03-19:
It seems to work perfectly with the Teensy LC, just seems to be having issues with The Teensy 4.0

MatePlaque, 2024-05-06:
can someone help with this situation were i am getting this when i try to run the above code to the flash chip its just saying 
 Previous data found at address 0
 You must fully erase the chip before this test
 found this: 00 00 00 00 0A FA CA 25 
 correct: 00 00 00 00 15 F5 95 4B 

Tests Failed :{

The flash chip may be left in an improper state.
You might need to power cycle to return to normal.

PaulStoffregen, 2024-05-06:
Try shorter wires.  SPI is very fast, especially on Teensy 4.0.