PaulStoffregen/SerialFlash #79: Arduino MKR MEM Shield - Wrong JEDEC ID - Used to work

sambartle, 2021-05-10:
### Description
Until recently this code worked fine to access the Winbond 25Q16JVIQ Flash chip on the MKR MEM Shield. 
It appears that changes (somewhere external) have now broken this code as it cannot communicate with this chip anymore. 
The JEDEC ID Returned is incorrect and the chip does not work.
I can reproduce this on 5 different sets of new MKR GSM1400 and MKR MEM Shields, and have redownloaded a clean Arduino IDE (run in portable mode for isolation) and new library download.
 

### Steps To Reproduce Problem
Download Arduino IDE. 
Connect MKR MEM Shield to Arduino MKR GSM 1400.
Install MKR Board and this library from library manager (or github result is same)
Open Example: RawHardWareTest.ino
Change FlashChipSelect from 6 to 5 (for the MKR MEM Shield)
Flash code and observer serial monitor.

### Hardware & Software

Board: Arduino MKR GSM 1400
Shields / modules used: Arduino MKR MEM Shield
Arduino IDE version: 1.8.13
Version info & package name: Arduino SAMD Boards (1.8.11)
Operating system & version: Windows 10 Pro 20H1

### Arduino Sketch
The included RawHardwareTest.ino with only 1 change.. 
```
const int FlashChipSelect = 5; // digital pin for flash chip CS pin
```

### Errors or Incorrect Output

> Raw SerialFlash Hardware Test
> 
> Read Chip Identification:
>   JEDEC ID:     F7 20 A
>   Part Number: (unknown chip)
>   Memory Size:  1048576 bytes
>   Block Size:   65536 bytes
> 
> Reading Chip...
> 
> Writing 512 signatures
>   error writing signature at 0
>   Read this: 00 00 00 00 0A FA CA 25 
>   Expected:  00 00 00 00 15 F5 95 4B 
> 
> Tests Failed  :{
> 
> The flash chip may be left in an improper state.
> You might need to power cycle to return to normal.

As you can see the JEDEC ID is Incorrect and the test fails.

etix, 2021-06-02:
> This is caused by a change in the SAMD core in arduino.. v1.8.10 and 1.8.9 work fine, but the latest 1.8.11 causes this error.

Confirmed, 1.8.10 works fine, 1.8.11 does not.
Tested on Mini Ultra Pro v3 (ATSAMD21G18A-AU).

PaulStoffregen, 2021-06-08:
Does this SAMD issue solve the problem?
https://github.com/arduino/ArduinoCore-samd/pull/622

Should I close this issue, or leave it open?

sambartle, 2021-06-08:
I haven't been able to test the changes as the signature file is missing in that issue, however if I make the changes manually it does fix the issue.

So I am certain it will and this can be closed.

sambartle, 2021-06-08:
Will be resolved in Arduino core. No issue here.

closed 2021-06-08 by sambartle

stefat, 2021-07-07:
Hi all, I am a newbie with and MKR1010 + MKR MEM. I became crazy for your this same issue. At least I now know you've found the cause. I then tried to install 1.8.10 in my Mac (already running 1.8.15 IDE) trying to fix the speed issue with the SPI but the older IDE does not run. Any clue ? How can I run the older 1.8.10 in my Mac ? Or.. how can I fix this issue ? Thank you to anybody spending some time to help.

PaulStoffregen, 2021-07-07:
Best to ask for help on the Arduino forum.

stefat, 2021-07-07:
ok thanks

stefat, 2021-07-07:
BTW: I've just done this  
....
bool ret = SerialFlash.begin(chipSelect);
SPI.setClockDivider(SPI_CLOCK_DIV64);
....
and it now works.

sdrshnptl, 2022-03-29:
> SPI.setClockDivider(SPI_CLOCK_DIV64);

This solve my problem on ESP32 with salvaged W25Q80 FLASH
testing further!
[image]