PaulStoffregen/SerialFlash #86: Incompatible board: Rocket Scream Mini Ultra Pro V2 (With Radio)

ianaraphael, 2022-02-03:
### Description

Describe your problem.
Board fails SerialFlash hardware test. Additionally, I am unable to create new files in flash memory despite SerialFlash.begin(); returning true and the appearance of successful communication with the flash chip.

### Steps To Reproduce Problem

Please give detailed instructions needed for anyone to attempt to reproduce the problem.

### Hardware & Software

_Board_
Rocket Scream Mini Ultra Pro V2 (DEV-00064)
_Shields / modules used_
None
_Arduino IDE version_
1.8.15
_Teensyduino version (if using Teensy)
Version info & package name (from Tools > Boards > Board Manager)_
Arduino AVR Boards 1.8.3
_Operating system & version_
macOS Catalina 10.15.7
_Any other software or hardware?_

### Arduino Sketch

```cpp
// Change the code below by your sketch (please try to give the smallest code which demonstrates the problem)
#include <Arduino.h>

// libraries: give links/details so anyone can compile your code for the same result

// RawHardwareTest - Check if a SPI Flash chip is compatible
// with SerialFlash by performing many read and write tests
// to its memory.
//
// The chip should be fully erased before running this test.
// Use the EraseEverything to do a (slow) full chip erase.
//
// Normally you should NOT access the flash memory directly,
// as this test program does.  You should create files and
// read and write the files.  File creation allocates space
// with program & erase boundaries within the chip, to allow
// reading from any other files while a file is busy writing
// or erasing (if created as erasable).
//
// If you discover an incompatible chip, please report it here:
// https://github.com/PaulStoffregen/SerialFlash/issues
// You MUST post the complete output of this program, and
// the exact part number and manufacturer of the chip.

#include <SerialFlash.h>
#include <SPI.h>

const int FlashChipSelect = 4; // digital pin for flash chip CS pin
//const int FlashChipSelect = 21; // Arduino 101 built-in SPI Flash

#define Serial SerialUSB

SerialFlashFile file;

const unsigned long testIncrement = 4096;

void setup() {

  //uncomment these if using Teensy audio shield
  //SPI.setSCK(14);  // Audio shield has SCK on pin 14
  //SPI.setMOSI(7);  // Audio shield has MOSI on pin 7

  //uncomment these if you have other SPI chips connected
[... 453 lines cut ...]
  Serial.println();
}

### Errors or Incorrect Output

If you see any errors or incorrect output, please show it here.  Please use copy & paste to give an exact copy of the message.  Details matter, so please show (not merely describe) the actual message or error exactly as it appears.

Raw SerialFlash Hardware Test

Read Chip Identification:
  JEDEC ID:     C1 FB 1
  Part Nummber: (unknown chip)
  Memory Size:  1048576 bytes
  Block Size:   65536 bytes

Reading Chip...
  Previous data found at address 0
  You must fully erase the chip before this test
  found this: 09 12 16 00 01 00 02 11 
     correct: 00 00 00 00 15 F5 95 4B 

Tests Failed  :{

The flash chip may be left in an improper state.
You might need to power cycle to return to normal.

FrankBoesing, 2022-02-04:
As it looks, this library is recommended by the manufacturer. It seems very strange that the library is recommended untested. One should therefore assume that it has been tested.

It probably makes more sense to ask the manufacturer of the board.

[https://www.rocketscream.com/blog/product/mini-ultra-pro-v2-with-radio/](https://www.rocketscream.com/blog/product/mini-ultra-pro-v2-with-radio/)

> On-board **2 MB (16 Mbit) serial flash** for data logging: 
> 
> *   [SerialFlash library](https://github.com/PaulStoffregen/SerialFlash) by [Paul Stoffregen](https://www.pjrc.com/) is highly recommended to access the on-board serial flash

rocketscream, 2023-06-25:
Sorry if this is late. This is mentioned on the board guide on the need to deselect the radio (pin D5) as they are both on the same SPI bus. All board were tested using this library as well. @PaulStoffregen please close this issue and my apologies.

hokascha, 2023-12-01:
Even after pulling up D5 the hardware test cannot write. SerialFlash.ready() never returns true. Added some serial logging:

```
Serial.println();
    Serial.print("Writing ");
    Serial.print(count);
    Serial.println(" signatures");
    memset(buf, 0, sizeof(buf));
    memset(sig, 0, sizeof(sig));
    memset(buf2, 0, sizeof(buf2));
    address = 0;
    first = true;
    while (address < chipsize) {
      SerialFlash.read(address, buf, 8);
      if (is_erased(buf, 8)) {
        create_signature(address, sig);
        Serial.print("write to ");
        Serial.println(address);
        //Serial.printf("write %08X: data: ", address);
        //printbuf(sig, 8);
        SerialFlash.write(address, sig, 8);
        Serial.println("waiting...");
        while (!SerialFlash.ready()) ; // wait
        Serial.println("reading...");
        SerialFlash.read(address, buf, 8);
[...]
```
Output:
```
14:54:05.724 -> Raw SerialFlash Hardware Test
14:54:05.724 -> 
14:54:05.724 -> Read Chip Identification:
14:54:05.724 ->   JEDEC ID:     FF FF FF
14:54:05.724 ->   Part Nummber: (unknown chip)
14:54:05.724 ->   Memory Size:  1048576 bytes
14:54:05.724 ->   Block Size:   65536 bytes
14:54:05.724 -> 
14:54:05.724 -> Reading Chip...
14:54:05.757 -> ready
14:54:05.757 -> 
14:54:05.757 -> Writing 512 signatures
14:54:05.757 -> write to 0
14:54:05.757 -> waiting...
```

any idea?

hokascha, 2023-12-05:
@rocketscream tried in two different Mini Ultra Pro boards without success. Can you help?

hokascha, 2023-12-07:
Looks like power cycling the board helps! Now the RawHardwareTest runs succesfully.

1. Erase everything
2. Power cycle
3. RawHardwareTest